celsworth/lxp-bridge #235: EG4 6000XO (rebranded luxpower) Support

iannecj, 2024-02-15:
**Enhancement request**

I have LXP working in Home Assistant against this new very popular off grid all in one inverter the EG4 600XP which is made by lux

[https://eg4electronics.com/backend/wp-content/uploads/2023/10/EG4-6000XP-Manual.pdf](url)

It has this serial modbus protocol file and we speak to it through a lux wifi dongle which must be larger similar to other lux units as the bridge shows much of this data.

[https://eg4electronics.com/resources/downloads#:~:text=EG4%C2%AE%206000XP%20%2D%20MODBUS%20Communication%20Protocol.pdf](url)

I do notice generator data is not bridged to MQTT.  Also I see unknown in a number of fields I can read through app in particular status, fault code and warning code.

Question:  Will you support this unit?  I expect it to be one of the most common variants in the wild.

Thanks for your great work regardless.

pmccut, 2024-02-26:
I was looking for the generator (ac-coupled solar) data as well for my eg4 18k. I found the modbus pdf posted on a forum and the generator uses register 120+ which this project does not yet consume. I've got mine decoding it locally and can send out a patch later in the week to add a ReadInput4 to see if that works for the 6000XP as well.

pmccut, 2024-03-03:
https://github.com/celsworth/lxp-bridge/pull/239 for the generator data if you can try this on your EG4. I'm not looking at this via HA - just a local prometheus instance so not sure if there's something else needed in HA.

celsworth, 2024-03-04:
Now merged into master - yes we need a bit more to get them visible in Home Assistant, they need adding in https://github.com/celsworth/lxp-bridge/blob/master/src/home_assistant.rs

Feel free to give this a go, or I can  take a look in a few days.