celsworth/lxp-bridge #238: Occasional Network Disconnect

JohnMacrae, 2024-02-22:
Hi Chris, 

Whilst I realise this isn't a problem for most (and probably not really your problem at all), I'm facing an interesting intermittent issue where for some (undetermined) reason, HA loses the MQTT connection with lxp-bridge. Both are running in separate containers with the MQTT server on a separate machine. I can restore full functionality only by restarting the lxp container.

My initial analysis is that there is possibly an intermittent network problem. When it occurs, lxp 'disappears' from HA but all other MQTT devices remain addressable.  I'm wondering if there is a simple 'fix' to the 'dropped' connection?

jacovt, 2024-03-13:
Hello @JohnMacrae ,

Sorry about he delay here - could you perhaps provide some logs from your side when this happens? Both HA and lxp-bridge.

In your scenatio both HA and lxp-bridge would be clients connecting to the external MQTT server, so it would be interesting to see where things break down. Judging by your explanation, it looks like it might be breaking down between lxp-bridge and the MQTT server (you are still seeing other MQTT things on HA).

It might also be a connection break between lxp-bridge and the inverters, so the logs might help show if that is the case.

---Jaco

JohnMacrae, 2024-03-13:
Hi Jaco,

Sadly I don't have the old logs. I've also now moved the MQTT server to be on the same machine as LXP Bridge and HA and the problem has not recurred. I consider this an 'edge' case as most people will be running the MQTT server on the same machine as lxp-bridge and HA - and most people probably have a better network than me :)

I did notice when it last happened that the connection between lxp-bridge and the inverter was operating (although it may have stopped and restarted - but the MQTT connection was dead.) I'm not sure the problem deserves too much attention but maybe worth keeping in the back of your mind if you're in there taking a look.

Sadly, I'm no Rust programmer otherwise I'd already have been there.