celsworth/lxp-bridge #255: Latest 0.14.0-dev (7852b00) connects to MySQL/Influx but does not add any new rows

zatie12, 2024-03-14:
Hello,

I was testing the latest build and noticed that despite successfully connecting to MySQL, lxp-bridge doesn't appear to be adding any new rows to the mysql or influx tables. I've checked the logs and there are no errors reported, only a successful connect to both servers respectively:

Influx:
lxp-bridge[3288200]: [2024-03-14T08:45:56.939 INFO lxp_bridge::influx] initializing influx at http://localhost:8086

MySQL:
lxp-bridge[3288200]: [2024-03-14T08:45:56.938 INFO lxp_bridge::database] initializing database
lxp-bridge[3288200]: [2024-03-14T08:45:56.980 INFO lxp_bridge::database] database connected

Those are the only reported lines relating to Influx or MySQL in the journal that I have been able to find.

Thanks very much, hope this helps!
Jez

celsworth, 2024-03-14:
Yeah this is known unfortunately, I'm in the middle of rewriting a lot of things to fix this. The short version is a PR was merged that makes lxp-bridge wait for a 4th input data packet from the inverter before updating various things. Only some inverters send this, so if yours doesn't, it will never trigger the influx/db writes.

There's a bit list of things I'm working on at https://github.com/celsworth/lxp-bridge/issues/245 to try and fix this properly, the PR is at https://github.com/celsworth/lxp-bridge/pull/249 but it's still WIP so not ready for a test yet.

Best thing is just to revert to previous build for now.

zatie12, 2024-03-14:
That would explain it, I guess my inverter doesn't send that 4th packet.

No problem at all, I've reverted for now. Thanks for the reply. I'll keep an eye, always happy to test.

celsworth, 2024-03-14:
If you're interested and in a position to run a binary rather than a Docker container, the builds for #245 should now be working for InfluxDB again (but not databases) - at bottom of https://github.com/celsworth/lxp-bridge/actions/runs/8284643518

May be a few more days/week or two before this makes into `master` and thus dev Docker build. And I'm not quite sure if databases will be working again in that, I'll have a go. But they're a bit of a pain :(

zatie12, 2024-03-14:
I ran the linux-amd64 build from your linked page on my server here in place of the known-good build. It runs fine but doesn't appear to update Influx. 2 lines spotted from a brief run below (looks like it's trying to write, not sure why I didn't see an update in Grafana though, will have to investigate further tomorrow).

```
[2024-03-14T22:09:37.131 INFO lxp_bridge::influx] initializing influx at http://localhost:8086

[2024-03-14T22:09:44.901 DEBUG lxp_bridge::influx] [Line { measurement: Measurement("inputs"), fields: {FieldName("p_rec"): Integer(0), FieldName("min_cell_temp"): Float(15.3), FieldName("p_pv_1"): Integer(2), FieldName("v_ac_s"): Float(128.1), FieldName("v_ac_t"): Float(0.0), FieldName("charge_volt_ref"): Float(53.2), FieldName("bat_capacity"): Integer(250), FieldName("e_eps_all"): Float(1.5), FieldName("p_eps"): Integer(0), FieldName("e_pv_all_2"): Float(0.0), FieldName("bat_current"): Float(0.0), FieldName("max_dischg_curr"): Float(125.0), FieldName("e_to_user_all"): Float(4485.4), FieldName("e_dischg_day"): Float(4.6), FieldName("v_bus_2"): Float(316.5), FieldName("v_pv_2"): Float(0.0), FieldName("e_chg_day"): Float(4.7), FieldName("status"): String("Standby"), FieldName("s_eps"): Integer(0), FieldName("t_rad_1"): Integer(22), FieldName("vbat_inv"): Float(49.4), FieldName("fault_code"): String("OK"), FieldName("p_pv_2"): Integer(0), FieldName("p_charge"): Integer(0), FieldName("e_to_grid_all"): Float(7096.3), FieldName("v_ac_r"): Float(243.1), FieldName("e_chg_all"): Float(4949.4), FieldName("v_bus_1"): Float(373.7), FieldName("f_ac"): Float(49.94), FieldName("e_inv_all"): Float(4374.5), FieldName("soh"): Integer(97), FieldName("e_dischg_all"): Float(4294.3), FieldName("cycle_count"): Integer(0), FieldName("f_eps"): Float(49.94), FieldName("p_grid"): Integer(833), FieldName("v_bat"): Float(49.4), FieldName("bat_count"): Integer(5), FieldName("warning_code"): String("OK"), FieldName("e_pv_day_3"): Float(0.0), FieldName("runtime"): Integer(62884614), FieldName("e_to_user_day"): Float(5.8), FieldName("dischg_cut_volt"): Float(40.0), FieldName("min_cell_voltage"): Float(0.45), FieldName("v_eps_r"): Float(243.1), FieldName("v_pv_1"): Float(0.0), FieldName("p_pv"): Integer(2), FieldName("e_pv_day_2"): Float(0.0), FieldName("e_eps_day"): Float(0.0), FieldName("pf"): Float(1.0), FieldName("p_to_user"): Integer(833), FieldName("max_chg_curr"): Float(125.0), FieldName("e_pv_all_3"): Float(0.0), FieldName("t_rad_2"): Integer(19), FieldName("e_rec_all"): Float(4896.9), FieldName("p_inv"): Integer(0), FieldName("p_pv_3"): Integer(0), FieldName("p_battery"): Integer(0), FieldName("e_pv_day_1"): Float(20.1), FieldName("t_inner"): Integer(20), FieldName("e_pv_all_1"): Float(17833.7), FieldName("soc"): Integer(60), FieldName("max_cell_voltage"): Float(1.25), FieldName("e_inv_day"): Float(4.8), FieldName("v_eps_t"): Float(1235.2), FieldName("e_rec_day"): Float(4.5), FieldName("v_eps_s"): Float(307.2), FieldName("p_to_grid"): Integer(0), FieldName("e_to_grid_day"): Float(7.7), FieldName("t_bat"): Integer(0), FieldName("v_pv_3"): Float(0.2), FieldName("max_cell_temp"): Float(15.3), FieldName("p_discharge"): Integer(0)}, tags: {TagName("datalog"): TagValue("BAxxxxxxxxx")}, timestamp: None }]
```

celsworth, 2024-03-14:
Oh that's a bit odd, if it got to that debug line then the next line is to push it up to Influx (and if that fails it logs an error) so would have thought that'd work.

Be interested to see if there's any failure logs from Influx. I did get some new entries in my own InfluxDB using this so I think it should work.

zatie12, 2024-03-14:
I had a quick look with trace and query logging enabled in Influx config. I'm seeing loads of these lines from lxp-bridge but Influx doesn't appear to be recording any insert queries for the same timestamp as the long example in my previous post, unless I need to enable further debugging options in Influx :-)

lxp-bridge[3351011]: [2024-03-14T22:34:09.531 ERROR lxp_bridge::influx] push failed: Unknown - retrying in 10s
lxp-bridge[3351011]: [2024-03-14T22:34:19.535 ERROR lxp_bridge::influx] push failed: Unknown - retrying in 10s
....
....

celsworth, 2024-03-15:
"Unknown" isn't too useful :( The only thing I changed here is I don't explicitly send a timestamp to Influx anymore, but it should use the current time in that case. Maybe thats changed in different versions, which Influx are you using?

Anyway I'll add it back in and we can see if that helps.

celsworth, 2024-03-15:
Any change with the builds at https://github.com/celsworth/lxp-bridge/actions/runs/8292569524 ?

zatie12, 2024-03-15:
I would agree that I didn't think specifying the timestamp is necessary, I'm sure I don't do that with some other ingestions but I guess I'm wrong on that one.

Version is 1.6.7~rc0

It's the version that comes with Ubuntu 23.10. I'll upgrade this system back to LTS when 24.04 releases but it looks like the Influx version will remain the same from the package search database.

celsworth, 2024-03-15:
Yeah same version as me. Ignore previous deleted message with new builds, realised it wasn't actually adding the time doh. This one should add it: https://github.com/celsworth/lxp-bridge/actions/runs/8292662202

celsworth, 2024-03-15:
As a test could you try a new database? No need to wipe your old one, but changing lxp-bridge config `influx.database` to `lxp2` or anything (and creating that) and seeing if it adds rows there?

zatie12, 2024-03-15:
Appears to be a similar issue with #249 

lxp-bridge[1687129]: [2024-03-15T07:23:26.686 ERROR lxp_bridge::influx] push failed: Unknown - retrying in 10s

I'll work on that new database now and report back

celsworth, 2024-03-15:
If a new database works I wonder if it's because some columns have changed type - for example status used to be an int, but is now a string. So maybe it's getting upset with that, but not sure why it worked for me..

zatie12, 2024-03-15:
Some good news - it is successfully adding new rows to a fresh database, lxp2

celsworth, 2024-03-15:
Ok, just have to work out what breaks it, my guess is status. I wonder if mine worked because I had no rows for about a year so for internal Influx reasons it didn't mind the type changing. Bit weird..

celsworth, 2024-03-15:
One last test for now, if you could try the binaries at https://github.com/celsworth/lxp-bridge/actions/runs/8292872863 when they're built.

This one hardcodes the status field to 0, so it's not a proper fix, but it does test my theory about whether the only issue is changing the status field from an int to a string.

Try this on your existing database if you can (or a copy of it)

zatie12, 2024-03-15:
Sure, I'll try that now. Not sure if this is useful in the meanwhile - this is the existing inputs table:

```
> SHOW FIELD KEYS ON "lxp" FROM "inputs"
name: inputs
fieldKey            fieldType
--------            ---------
bat_capacity        integer
bat_count           integer
bat_current         float
bat_status_0        integer
bat_status_1        integer
bat_status_2        integer
bat_status_3        integer
bat_status_4        integer
bat_status_5        integer
bat_status_6        integer
bat_status_7        integer
bat_status_8        integer
bat_status_9        integer
bat_status_inv      integer
bms_event_1         integer
bms_event_2         integer
bms_fw_update_state integer
charge_volt_ref     float
cycle_count         integer
dischg_cut_volt     float
e_chg_all           float
e_chg_day           float
e_dischg_all        float
e_dischg_day        float
e_eps_all           float
e_eps_day           float
e_inv_all           float
e_inv_day           float
e_pv_all            float
e_pv_all_1          float
e_pv_all_2          float
e_pv_all_3          float
e_pv_day            float
e_pv_day_1          float
e_pv_day_2          float
e_pv_day_3          float
e_rec_all           float
e_rec_day           float
e_to_grid_all       float
e_to_grid_day       float
e_to_user_all       float
e_to_user_day       float
f_ac                float
f_eps               float
fault_code          integer
internal_fault      integer
max_cell_temp       float
max_cell_voltage    float
max_chg_curr        float
max_dischg_curr     float
min_cell_temp       float
min_cell_voltage    float
p_battery           integer
p_charge            integer
p_discharge         integer
p_eps               integer
p_grid              integer
p_inv               integer
p_pv                integer
p_pv_1              integer
p_pv_2              integer
p_pv_3              integer
p_rec               integer
p_to_grid           integer
p_to_user           integer
pf                  float
runtime             integer
s_eps               integer
soc                 integer
soh                 integer
status              integer
t_bat               integer
t_inner             integer
t_rad_1             integer
t_rad_2             integer
v_ac_r              float
v_ac_s              float
v_ac_t              float
v_bat               float
v_bus_1             float
v_bus_2             float
v_eps_r             float
v_eps_s             float
v_eps_t             float
v_pv                float
v_pv_1              float
v_pv_2              float
v_pv_3              float
vbat_inv            float
warning_code        integer
```

celsworth, 2024-03-15:
Oh fault_code and warning_code are probably going to suffer the same fate.

Ok I need more of a rethink :) Going to have to work out how to get the original ints back into Influx

celsworth, 2024-03-15:
https://github.com/celsworth/lxp-bridge/actions/runs/8292922880 will also bodge out fault_code and warning_code

zatie12, 2024-03-15:
Sadly still the same issue with your latest

lxp-bridge[1822715]: [2024-03-15T07:50:51.789 ERROR lxp_bridge::influx] push failed: Unknown - retrying in 10s

Trying the next one now :-)

zatie12, 2024-03-15:
That works!

celsworth, 2024-03-15:
Awesome thanks, that at least confirms a goal to work towards :)

zatie12, 2024-03-15:
Pleasure :-)

celsworth, 2024-03-15:
https://github.com/celsworth/lxp-bridge/actions/runs/8293129761 has a better fix and should send proper status/fault_code/warning_code to InfluxDb again if you could test?

Thanks!

zatie12, 2024-03-15:
That works and I'm seeing the correct status code (charging)

celsworth, 2024-03-15:
I'll leave this open as while we solved one problem, there's still missing database integration ;)

zatie12, 2024-03-15:
Sounds good :)

zatie12, 2024-03-15:
I'll pop the MySQL show create table here just in case that comes in handy as well:

```
| inputs | CREATE TABLE `inputs` (
  `id` int NOT NULL AUTO_INCREMENT,
  `status` int NOT NULL,
  `v_pv_1` double NOT NULL,
  `v_pv_2` double NOT NULL,
  `v_pv_3` double NOT NULL,
  `v_bat` double NOT NULL,
  `soc` int NOT NULL,
  `soh` int NOT NULL,
  `p_pv` int NOT NULL,
  `p_pv_1` int NOT NULL,
  `p_pv_2` int NOT NULL,
  `p_pv_3` int NOT NULL,
  `p_charge` int NOT NULL,
  `p_discharge` int NOT NULL,
  `v_ac_r` double NOT NULL,
  `v_ac_s` double NOT NULL,
  `v_ac_t` double NOT NULL,
  `f_ac` double NOT NULL,
  `p_inv` int NOT NULL,
  `p_rec` int NOT NULL,
  `pf` double NOT NULL,
  `v_eps_r` double NOT NULL,
  `v_eps_s` double NOT NULL,
  `v_eps_t` double NOT NULL,
  `f_eps` double NOT NULL,
  `p_to_grid` int NOT NULL,
  `p_to_user` int NOT NULL,
  `e_pv_day` double NOT NULL,
  `e_pv_day_1` double NOT NULL,
  `e_pv_day_2` double NOT NULL,
  `e_pv_day_3` double NOT NULL,
  `e_inv_day` double NOT NULL,
  `e_rec_day` double NOT NULL,
  `e_chg_day` double NOT NULL,
  `e_dischg_day` double NOT NULL,
  `e_eps_day` double NOT NULL,
  `e_to_grid_day` double NOT NULL,
  `e_to_user_day` double NOT NULL,
  `v_bus_1` double NOT NULL,
  `v_bus_2` double NOT NULL,
  `e_pv_all` double NOT NULL,
  `e_pv_all_1` double NOT NULL,
  `e_pv_all_2` double NOT NULL,
  `e_pv_all_3` double NOT NULL,
  `e_inv_all` double NOT NULL,
  `e_rec_all` double NOT NULL,
  `e_chg_all` double NOT NULL,
  `e_dischg_all` double NOT NULL,
  `e_eps_all` double NOT NULL,
  `e_to_grid_all` double NOT NULL,
  `e_to_user_all` double NOT NULL,
  `t_inner` int NOT NULL,
  `t_rad_1` int NOT NULL,
  `t_rad_2` int NOT NULL,
  `t_bat` int NOT NULL,
  `runtime` int NOT NULL,
  `max_chg_curr` double NOT NULL,
  `max_dischg_curr` double NOT NULL,
  `charge_volt_ref` double NOT NULL,
  `dischg_cut_volt` double NOT NULL,
  `bat_status_0` int NOT NULL,
  `bat_status_1` int NOT NULL,
  `bat_status_2` int NOT NULL,
  `bat_status_3` int NOT NULL,
  `bat_status_4` int NOT NULL,
  `bat_status_5` int NOT NULL,
  `bat_status_6` int NOT NULL,
  `bat_status_7` int NOT NULL,
  `bat_status_8` int NOT NULL,
  `bat_status_9` int NOT NULL,
  `bat_status_inv` int NOT NULL,
  `bat_count` int NOT NULL,
  `bat_capacity` int NOT NULL,
  `datalog` varchar(10) CHARACTER SET utf8mb4 COLLATE utf8mb4_0900_ai_ci NOT NULL,
  `created_at` timestamp NOT NULL,
  `bat_current` double NOT NULL DEFAULT '0',
  `bms_event_1` int NOT NULL DEFAULT '0',
  `bms_event_2` int NOT NULL DEFAULT '0',
  `max_cell_voltage` double NOT NULL DEFAULT '0',
  `min_cell_voltage` double NOT NULL DEFAULT '0',
  `max_cell_temp` double NOT NULL DEFAULT '0',
  `min_cell_temp` double NOT NULL DEFAULT '0',
  `bms_fw_update_state` int NOT NULL DEFAULT '0',
  `cycle_count` int NOT NULL DEFAULT '0',
  `vbat_inv` double NOT NULL DEFAULT '0',
  `p_eps` int NOT NULL DEFAULT '0',
  `s_eps` int NOT NULL DEFAULT '0',
  `p_battery` int NOT NULL DEFAULT '0',
  `p_grid` int NOT NULL DEFAULT '0',
  `internal_fault` int NOT NULL DEFAULT '0',
  `fault_code` int NOT NULL DEFAULT '0',
  `warning_code` int NOT NULL DEFAULT '0',
  PRIMARY KEY (`id`),
  KEY `index_created_at_datalog` (`created_at`,`datalog`) USING BTREE
) ENGINE=InnoDB AUTO_INCREMENT=3700376 DEFAULT CHARSET=utf8mb4 COLLATE=utf8mb4_0900_ai_ci |

```